gandalf15/HX711 #1: Help me please

Heavenwalker, 2017-10-21:
[results.txt](https://github.com/gandalf15/HX711/files/1403749/results.txt)
I am trying to create a weight scale using hx711 along a raspberry pi. I can confirm that both HX711 and the load censor are fully functional. Yet all I am getting are either false or random values. The results range from 

> Ready to use
> invalid data
> invalid data

to
  
> Ready to use
> invalid data
> Data subtracted by offset but still not converted to any unit: -0.19999999999999998
> Put known weight on the scale and then press Enter
> Traceback (most recent call last):
>   File "example.py", line 77, in <module>
>     raise ValueError('Cannot calculate mean value. Try debug mode.')
> ValueError: Cannot calculate mean value. Try debug mode.

As well as the contents of the attached txt.

Any idea what i might be doing wrong? I'm also attaching pictures of my setup. I have experimented with various pin places but nothing changed

Thank you.

[image]
[image]

gandalf15, 2017-10-24:
Hi,
I will try to help you. I noticed that you swapped clk and data pins. Notice in the example.py that data pin is 21 and CLK is 20. On the picture you uploaded I see that yellow wire is CLK going to pin 21 and gray which is data out should be in pin 21.
Let me know if that helps. :)

closed 2017-10-30 by gandalf15

layaannj, 2018-08-14:
Hi,
 I am having this same issue.But connection are correct .
I enabled debug mode. The output is -

/home/pi/Desktop/load_cell/hx711.py:33: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self._pd_sck, GPIO.OUT)	# pin _pd_sck is output only
/home/pi/Desktop/load_cell/hx711.py:34: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self._dout, GPIO.IN)		# pin _dout is input only
Not enough fast while reading data
Time elapsed: 9.21379999851979e-05
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

not ready
Not enough fast while reading data
Time elapsed: 6.786599999486498e-05
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

invalid data
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Not enough fast while reading data
Time elapsed: 0.00012052299999254501
Not enough fast while reading data
Time elapsed: 0.00010255399999437032
zero() got False back.

Not enough fast while reading data
Time elapsed: 7.338700001469078e-05
Not enough fast while reading data
Time elapsed: 0.00011651299999471121
Binary value as it has come: 0b0

Converted 2's complement value: 0

Not enough fast while reading data
Time elapsed: 0.0003099540000164325
Not enough fast while reading data
Time elapsed: 0.00011906399998906636
Not enough fast while reading data
Time elapsed: 0.00013786700000650853
Not enough fast while reading data
Time elapsed: 0.0001701069999739957
Not enough fast while reading data
Time elapsed: 6.437599998321275e-05
Not enough fast while reading data
Time elapsed: 0.0001499500000079479
Binary value as it has come: 0b0

Converted 2's complement value: 0

invalid data
Put known weight on the scale and then press Enter
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Not enough fast while reading data
Time elapsed: 0.00013552199999367076
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Traceback (most recent call last):
  File "/home/pi/Desktop/load_cell/load1.py", line 74, in <module>
    raise ValueError('Cannot calculate mean value. Try debug mode.')
ValueError: Cannot calculate mean value. Try debug mode.

Please help me?

Thank you

gandalf15, 2018-08-16:
Hi @layaannj,
This does not seem to be the same issue.
You have warning there that your GPIOs are in use.
I have a few questions:
1. Are you using Raspeberry Pi 2 or 3?
2. Are you running any other program that is using the GPIO? 
3. If you restart the Raspberry Pi, do you still get the warning about GPIO?
4. Are you running any other heavy processes? if yes, try to increase niceness: https://www.lifewire.com/uses-of-commands-nice-renice-2201087

Let me know.
Thanks
Marcel