gandalf15/HX711 #4: Python2.7 example has Python 3 hashbang

pearson, 2018-05-11:
Hello. I'm starting to use your library, and the first thing I ran into was that the Python 2.7 example has the Python 3 hashbang at the start:
`#!/usr/bin/env python3`

Yes, I should just convert myself to Python 3, especially for a new project... :-) I thought it was worth mentioning, though.

HeMe2, 2018-05-14:
I corrected this in my fork an created a pull request. 

Thank You for mentioning it.

gandalf15, 2018-05-14:
It is solved now. Closing this issue.

closed 2018-05-14 by gandalf15

pearson, 2018-05-14:
@HeMe2 , you're welcome! To be honest, that made me look into what's new in Python, and to realize that the end-of-life for Python 2.7.x is only two years away. It's time to (finally!) make the switch to Python 3. 

So, for me, it was a "good" bug. :-)