gandalf15/HX711 #6: [QUESTIONS] Using more than one HX711 board simultaneously

rafaellehmkuhl, 2018-07-07:
Hi! I have a project that uses 5 HX711 boards (each one reading one load cell) running simultaneously. I'm currently using an Arduino (with [this](https://github.com/olkal/HX711_ADC) library) to read from the boards, and them I send the data through serial to a RaspberryPy running python code, but I would rather do all the aquisition and processing on the RaspberryPi, without an Arduino in the middle.

Have you tested your library with more than one HX711 board simultaneously?

HeMe2, 2018-07-08:
I have no experience with the usage of more than one hx711. Since the used pins are set in the constructor of the HX711 object, it should work fine with more than one device. Given that you use different pins for each.

gandalf15, 2018-07-08:
Hi, I am using only two hx711 connected to a single Raspberry Pi2. There is no problem.  You just have to read them consecutively. That means using a single process that is trying access GPIOs. However you mentioned RaspberryPi and no specific version. Currently, I am working with Pi Zero W where I am having troubles to read correct output. So, I would recommend using Pi2 or Pi3 for now. Soon, I'll create a new dev branch with the latest updates.

rafaellehmkuhl, 2018-07-08:
Nice, that was why I asked. The HX711 chip needs a good timed aquisition, so I thought that was the problem why it is easy to make many work on an Arduino but not on a Raspberry. Timing is the issue (more specifically the way they shutdown if you let the SCK pin high for too much time).

Do you know why are you having trouble with the Zero W? Maybe it is not aquisiting fast enough? My goal was to eventually change my project from a RaspberryPi3 + Arduino + HX711s to a RaspberryPi Zero W + RPiZW shield.

gandalf15, 2018-07-08:
I just guess that it is because Pi Zero has single core while Pi2 is quad-core. So, I think that sometimes my process is preempted while reading... You know the rest of the story.
I am working on resolving this issue. The solution may be increase niceness of the process with nice. However, I did not test it yet. I'll let you know my progress within a few days (2 - 5). Meanwhile, look at this repo: https://github.com/ggurov/hx711
He is using C. I can say that this works on Pi Zero perfectly. 
Note: In his code you will see 4 x increment useless variable as ```b++```. Do not touch it it is important for right timing.

closed 2018-07-20 by gandalf15

adriencarbonaro, 2019-01-28:
Hey @gandalf15, did you resolve the reading issue with a RPi Zero ? I'm using one right now and I found out that my readings were less precise than the other Pi.

adriencarbonaro, 2019-01-28:
Ahah I know C but unfortunately I don't have time to implement it in C.
I will look into it in my free time after this project! Thank you!