gandalf15/HX711 #7: SyntaxError: unexpected EOF while parsing

hristakiev, 2018-11-15:
I try HX711_Python2.7 and got:
not ready
invalid data
invalid data
Put known weight on the scale and then press Enter
Traceback (most recent call last):
  File "./example.py", line 51, in <module>
    input('Put known weight on the scale and then press Enter')
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

gandalf15, 2018-11-25:
Hi, 
I would like to ask you which version of raspberry pi are you using? Did you check your wiring?
It seems that raspberry pi cannot communicate with the HX711. Please provide more details of your setup. I'll try to help you get it working.

hristakiev, 2018-11-26:
Hello,  I'm happy  that you answered me.
I use Raspberry Pi 3 Model B+, Python 2.7.13, Raspbian GNU/Linux 9 (stretch)
The problem only persist with your library, but works with https://github.com/tatobari/hx711py or https://github.com/KiwisInSpace/hx711loadcell so  wiring is not the problem.Tomorrow ill go to my lab and will try python 3.
Actually I'm looking for a solution for this:   https://github.com/KiwisInSpace/hx711loadcell/issues/1
I'll use for a bee hive measuring and need some last known weight recover  on reset ?

hristakiev, 2018-11-26:
sudo /usr/bin/python3.5 example.py 
not ready
invalid data
invalid data
Put known weight on the scale and then press Enter
Traceback (most recent call last):
  File "example.py", line 73, in <module>
    raise ValueError('Cannot calculate mean value. Try debug mode.')
ValueError: Cannot calculate mean value. Try debug mode

the same is as:  sudo ./example.py

gandalf15, 2018-11-26:
I think the problem is in my code. I used filter for removing wrong data... This was not the best solution because it is based on stdev...
Please try to checkout to dev branch. Then run the example with python3. [here](https://github.com/gandalf15/HX711/tree/dev/HX711_Python3) This should give you some output. Please report back.
There is no filter and I started to work on improvements. 
I am definitely interested in the problem you have "So i'm looking for a solution when there is a constant load on the cell and after power failure to restore last known weight?" and I can try to help you. I was also planing to work on this library since the next week Monday (end of Uni). 
If you are not in hurry, I'll solve the problems and implement something that saves the data in case of power failure. Just to be clear, power failure means that also the raspberry pi is restarted or only the HX711? Thanks

hristakiev, 2018-11-27:
Hi, I'm not in hurry for now I use https://github.com/KiwisInSpace/hx711loadcell with  Sqlite and UPS.
Yes, power failure means restart the hole raspberry pi.I tried https://github.com/gandalf15/HX711/tree/dev/HX711_Python3 and is the same:
sudo ./example.py 
not ready
invalid data
invalid data
Put known weight on the scale and then press Enter
Traceback (most recent call last):
  File "./example.py", line 77, in <module>
    raise ValueError('Cannot calculate mean value. Try debug mode.')
ValueError: Cannot calculate mean value. Try debug mode.

I just changed in example.py  #!/usr/bin/env python3 to #!/usr/bin/python3.5 and make it +x is it correct?

gandalf15, 2018-12-04:
Yes that looks correct. Can you please provide more details? For example, a detailed picture of wiring and description of pin numbers. Also, how you changed the line 11 in example.py. ```hx = HX711(dout_pin=21, pd_sck_pin=20, gain_channel_A=128, select_channel='B')```. Based on what you said it should be correct but we have to be sure. Then, try debug mode. Add to line 12 in example.py this statement ```hx.set_debug_mode(True)``` and paste it here. Thanks

gandalf15, 2018-12-05:
Try pull changes from master branch. now it should work. The reason is that your load cell is different way around. When you put the weight on the scale, the value decreases... But it should not be a problem now.

hristakiev, 2018-12-06:
Yes, now HX711_Python3 works :) , but HX711_Python2 not:
Ready to use
Raw data: -23013.0
Data subtracted by offset but still not converted to any unit: -0.0999999999985
Put known weight on the scale and then press Enter
Traceback (most recent call last):
  File "./example.py", line 51, in <module>
    input('Put known weight on the scale and then press Enter')
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing

gandalf15, 2018-12-06:
Which version of Python 2 are you using? It seems that the method `input()` has some problem. So it is not a problem with this library. If you try to use the HX711 in your project it should work without any problems now. So, I consider this issue for closed.
However, may I ask you to open a new issue (enhancement) and describe the problem with power loss and ability to remember the latest state before the power loss. Then you can track the progress.
Thanks.

closed 2018-12-06 by gandalf15