gandalf15/HX711 #10: Example program not working

Pudding01, 2018-12-03:
So I tried to run the example.py program (Python3 version) and I got this for an output:
not ready
invalid data
invalid data
Put known weight on the scale and then press Enter
Traceback (most recent call last):
File "example.py", line 73, in <module>
raise ValueError('Cannot calculate mean value. Try debug mode.")
ValueError: Cannot calculate mean value. Try debug mode.

So I slapped "hx.set_debug_mode(flag=True)" on line 12
Debug mode gave me:
Debugmode ENABLED
Binary value as it has come: 0b0

Converted 2's complement value: 0

Binary value as it has come: 0b0

Converted 2's complement value: 0

Not enough fast while reading data
Time elapsed: 0.00012099...
<several repetitions of this>
not ready
Not enough fast while reading data
Time elapsed: 6.099...e-05
Binary value as it has come: 0b0
...
invalid data
<another binary/convered 2's stuff being 0b0/0>

<more "Not enough fast"x3>
<keeps jumping back and forth between Binary/Converted 2's being 0b0/0 and Not enough fast with really small values>

invalid data

Put known weight on the scale and then press Enter
<more familiar stuff, no changes>
Traceback...

Pudding01, 2018-12-03:
Fortunately my CS professor had a spare full Raspberry Pi 3 to use, unfortunately when I moved the SD card over, moved the connections over, etc. and ran the example.py, same thing. Only difference is in the debug stuff. (less "Not enough fast", more 0/0b0 stuff).

So the problem might be more than not having enough cores...

gandalf15, 2018-12-03:
Please tell me how you connected the HX711. I cannot clearly see it on the picture. Are you using BCM numbering based on this diagram? https://pinout.xyz/pinout/io_pi_zero# for pi zero or this: https://pinout.xyz/# for pi3. And how do you create the class ```HX711(dout_pin=21, pd_sck_pin=20, gain_channel_A=128, select_channel='A')```
copy and paste your line here pls. Thanks

Pudding01, 2018-12-03:
The zero had ground to ground, VDD to 3.3V, VCC to 5V, DAT to 20, and CLK to 21. Same for the pi3

As for class creation, I literally took the example program, copied and pasted it over to the pi (same directory as the hx711.pi file), and tried to run it. (my code is identical to the example code, save for line 12 which I replaced the empty line with "hx.set_debug_mode(flag=True)"

Pudding01, 2018-12-03:
Looking over the wiring again I seem to have swapped DAT and CLK during the transfer...

Pudding01, 2018-12-03:
Well, good news is that it's working for the big pi, bad news is I feel like an idiot.

Thanks though, working now

Pudding01, 2018-12-03:
Anywho, I'll keep an eye out for the zero version of this so I can use my IoT project with my own pi instead of my professors. (It's going to allow you to see if a chicken has laid an egg without having to go outside first.)

gandalf15, 2018-12-03:
Cool stuff. I'm happy to help you. 
I am planing to rewrite the library during this Christmas holiday. So probably the next week I'll have the wrapper for HX711_C. Good luck.

closed 2018-12-03 by gandalf15

JoschuaGosda, 2020-03-21:
Hello gandalf15,
have you rewriten the library since this post? I am trying to use a raspberry pi zero w together with 4 load cells which are connected as a Wheatstone Bridge, following this blog: http://blog.medien.ifi.lmu.de/swh/2018/09/10/hx711-how-to-measure-incorrectly/
Unfortunately I can't get it work and always get the same error as Ata1362:
> **after running the Example.py, I received below message:**
> Traceback (most recent call last):
> File "/home/pi/Desktop/HX711/HX711_Python3/example.py", line 12, in
> err = hx.zero()
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 125, in zero
> result = self.get_raw_data_mean(readings)
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 446, in get_raw_data_mean
> filtered_data = self._data_filter(data_list)
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 680, in outliers_filter
> data_median = stat.median(data)
> File "/usr/lib/python3.5/statistics.py", line 353, in median
> raise StatisticsError("no median for empty data")
> statistics.StatisticsError: no median for empty data

I think the python code should also work for a raspberrypi pi zero, as the site (https://www.honey-pi.de/en/a-d-wandler-hx711/) are also using it without problems. Anyway, I would be glad to get it running the one way or another. :)
I also rechecked my connections but they seem to be fine...
Thanks in advance, best
Joschua

Amanda809, 2021-05-27:
Hello gandalf15 ,