gandalf15/HX711 #12: last known weight lost on power reset

hristakiev, 2018-12-07:
I use HX711 Python library which at  every hour write down in  sqlite3 weight and time.
INSERT INTO loadcellSample (weight, comment, sampleDateTime)
The problem is when there is a power reset and weight is still present on the scale the output goes back to 0.I have to remove the weight from the load cell,power reset and place the weight back on the scale.Problem is that my weight is 100Kgs and it is not practical to remove weight.
So i'm looking for a solution when there is a constant load on the cell and after power failure to restore last known weight?

gandalf15, 2019-01-01:
There you go mate.
https://github.com/gandalf15/HX711/blob/master/HX711_Python3/example_save_state.py

This is example that shows you how to do it. It cannot be part of the library. You do not have to save the state every time you read the load cell, you can use time or some kind of counter to control the writes to SD card.

gandalf15, 2019-01-01:
BTW. There is a new filter implemented of you use the latest version of the library. It improve reading by a lot.

hristakiev, 2019-01-01:
Happy new year, thank you don't forget me i get the idea to use a swap file, but when i power off the pi it create the swap_file.swp but its empty  and than:
sudo ./example_save_state.py /home/bee/work/save/HX711/HX711_Python3/hx711.py:60: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self._pd_sck, GPIO.OUT)  # pin _pd_sck is output only
Traceback (most recent call last):
  File "./example_save_state.py", line 18, in <module>
    hx = pickle.load(swap_file)
EOFError: Ran out of input

gandalf15, 2019-01-01:
Hi, Happy New Year to you too. I forgot to flush and fsynch. Please try now. It should work. It works on my table and I tried the situation with power failure. Also, you have to save the state only once at the beginning. Please let me know. Thanks.

hristakiev, 2019-01-02:
Thank you again, now everything works fine.Good job :)

gandalf15, 2019-01-03:
Glad to help.

closed 2019-01-03 by gandalf15