gandalf15/HX711 #16: [QUESTION] How to choose Channel + Gain ?

adriencarbonaro, 2019-02-03:
What is the difference between different `channel` ? How to choose it ?
And what is the best between different `gains` ?

Thank you very much.

gandalf15, 2019-02-04:
Hi,

this is from the data sheet: https://github.com/gandalf15/HX711/blob/master/Datasheet_hx711_english.pdf
 "The input multiplexer selects either Channel A
or B differential input to the low-noise
programmable gain amplifier (PGA). Channel A
can be programmed with a gain of 128 or 64,
corresponding to a full-scale differential input
voltage of ±20mV or ±40mV respectively, when
a 5V supply is connected to AVDD analog power
supply pin. Channel B has a fixed gain of 32."

if you look at this picture: https://github.com/gandalf15/HX711/blob/master/rate_pin_80sps.png
there are pins B- and B+. That is a place where you can connect second load cell. Just use the same E- and E+ pins for both load cells. Then you can select the channel from where to read the data with method "select_channel". Default gain for channel A is 128 and you can change the gain to 64 or back to 128 with method "set_gain_A" (So, it influences only channel A). Channel B has a single gain only 32.

adriencarbonaro, 2019-02-04:
Ok thanks ! And for a `HX711` I need to have `VCC` and `VDD` pinned to **5V**, am I right ?

gandalf15, 2019-02-04:
Majority of the time I use channel A with 128 gain. If you select gain 64, it also decreases the noise. Therefore, you may see more stable readings.

gandalf15, 2019-02-04:
> Ok thanks ! And for a `HX711` I need to have `VCC` and `VDD` pinned to **5V**, am I right ?

Yes. It takes quite flexible range of voltages from 2.7V up to 5.5V.

closed 2019-02-04 by gandalf15