gandalf15/HX711 #17: [QUESTION] How to install without a setup.py file

CohenP12, 2019-02-03:
Hi there,

Thanks very much for writing such a nice python 3 library for the HX711! I am very new to python 3, so I was wondering how I would actually install your library after it has been cloned. I am used to being able to run a command such as "sudo python setup.py install", but in this case there is no setup file.

So my question is, how do I actually install the library rather than just clone it.

Please let me know if you need clarification, sometimes when you're new to things (like I am) you're not good at asking good questions.

Cheers,

Cohen

gandalf15, 2019-02-04:
Hi Cohen,
no worries at all. When I started with programming I was also confused. Now, when I am more experienced I am still very often confused :smile: 
First, I would recommend reading this page for learning basic Python 3: https://www.tutorialspoint.com/python3
Try to go through every single part on the left side menu. 
To answer your question read this page: https://www.tutorialspoint.com/python3/python_modules.htm
Especially the section "Locating Modules" that says this:
"When you import a module, the Python interpreter searches for the module in the following sequences 

- The current directory.
- If the module is not found, Python then searches each directory in the shell variable PYTHONPATH.
- If all else fails, Python checks the default path. On UNIX, this default path is normally /usr/local/lib/python3/.

"
So, once you clone the repository to your Raspberry Pi, the example files will work just fine. If you want to import hx711 to your project (python file) that is not in this directory it will not work. 
Why?
Well, the example files, such as "example.py", work because they are in the same directory as hx711.py. Therefore, the python interpreter can find them.
In the second case, if you want to import hx711 to your project (python file) the python interpreter cannot find such file. 

The solution is to copy the file hx711.py to the directory you want to use it. If you want to use it on several places then copy it to the directory ```/usr/lib/python3/dist-packages/``` on your Raspberry Pi.

gandalf15, 2019-02-04:
If you have any question feel free to ask.

closed 2019-02-04 by gandalf15