gandalf15/HX711 #22: Error

Ata1362, 2019-04-08:
**after running the Example.py, I received below message:**
Traceback (most recent call last):
  File "/home/pi/Desktop/HX711/HX711_Python3/example.py", line 12, in <module>
    err = hx.zero()
  File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 125, in zero
    result = self.get_raw_data_mean(readings)
  File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 446, in get_raw_data_mean
    filtered_data = self._data_filter(data_list)
  File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 680, in outliers_filter
    data_median = stat.median(data)
  File "/usr/lib/python3.5/statistics.py", line 353, in median
    raise StatisticsError("no median for empty data")
statistics.StatisticsError: no median for empty data

gandalf15, 2019-04-11:
Hello @Ata1362 ,
As the error says "statistics.StatisticsError: no median for empty data" means that there is no data from HX711. There may be several reasons why this happened. 
1. check your connections (wiring) (you can upload picture here so we can check it together)
2. example.py assumes that your pins are "hx = HX711(dout_pin=21, pd_sck_pin=20)". If that is not the case change the pin numbers that it matches your connection.
3. Make sure you connect the HX711 to 5V
4. Do you have a good ground connection?

Let me know if that helps. If your problem still persists, upload a picture of your wiring (how exactly you connected your HX711 to Raspberry pi).

Ata1362, 2019-04-11:
Hello @gandalf15 
thank you for reply. 
There was no technical issue because I checked my code on python2 and it was working 100% OK.

gandalf15, 2019-04-12:
You are saying that "There was no..." so you already solved it? I do not really understand when you say that you checked your code with python2. This library should work only with python3 because it requires module "statistics".

Ata1362, 2019-04-14:
Hi @gandalf15 
Sorry if I did not explain it well. I wanted to use Python3 instead of Python2.
1. I have a Lib for Python2 which is working perfect with my hardware (so items 1-4 that you mentioned reagrding my circuit are not the problem in my case)
2. Yes, I solved my problem by using another HX711 Lib for Python 3. It is working on same Hardware.

gandalf15, 2019-04-14:
Thanks for letting me know. I do not know what was wrong.

closed 2019-04-14 by gandalf15

MBENKI, 2020-08-11:
Hello Gandalf hope everything's ok, i think i have the same problem this is what i get when i run in the terminal on my Raspberrypi 
sudo python3 example.py

Traceback (most recent call last):
File "example.py", line 12, in <module>
  err = hx.zero()
File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 125, in zero
  result = self.get_raw_data_mean(readings)
File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 446, in get_raw_data_mean
  filtered_data = self._data_filter(data_list)
File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 680, in outliers_filter
  data_median = stat.median(data)
File "/usr/lib/python3.7/statistics.py", line 379, in median
raise StatisticsError("no median for empty data")
statistics.StatisticsError: no median for empty data

What should i do, i am somewhat a beginner in this and i don't understand how to fix this . thank you so much.
[image]
[image]
[image]

xiMBest, 2020-11-15:
> Hello @gandalf15 hope everything's ok, i think i have the same problem this is what i get when i run in the terminal on my Raspberrypi
> sudo python3 example.py
> 
> Traceback (most recent call last):
> File "example.py", line 12, in
> err = hx.zero()
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 125, in zero
> result = self.get_raw_data_mean(readings)
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 446, in get_raw_data_mean
> filtered_data = self._data_filter(data_list)
> File "/home/pi/Desktop/HX711/HX711_Python3/hx711.py", line 680, in outliers_filter
> data_median = stat.median(data)
> File "/usr/lib/python3.7/statistics.py", line 379, in median
> raise StatisticsError("no median for empty data")
> statistics.StatisticsError: no median for empty data
> 
> What should i do, i am somewhat a beginner in this and i don't understand how to fix this . thank you so much.
> [image]
> [image]
> [image]

!!!!!!!!!!!!!!!!!!!!!HELP!!!!!!!!!!!!!!!!!
Hello check your pins `cause u connect to rpi to 5 and 6 pin and in code it`s 20 and 21 pin
if it`s okay my congratulation u:have the similar error as me
if someone knows the answer please will help me
!!!!!!!!!!!!!!!!!!!!!HELP!!!!!!!!!!!!!!!!!

MBENKI, 2020-11-15:
Hi @xiMBest just change the code like this :
hx = HX711(dout_pin=5, pd_sck_pin=6)
and yes it works just fine when you connect the rpi to pin 5 and 6