gandalf15/HX711 #23: What unit does get_raw_data_mean return?

iamhunter, 2019-05-07:
We are using the hx711 to just read strain rather than weight

gandalf15, 2019-05-07:
Hello,
you can use the method `get_raw_data_mean(readings=30)` to get just raw data (not converted and not subtracted by the offset) from HX711. 
The argument reading sets how many readings should read and then return average. If you do not want to even use the filter or average then just set the reading=1 and this will return immediately single reading.

closed 2019-05-07 by gandalf15

gandalf15, 2019-05-07:
Yes that is correct the value you get is unitless. You may want to use the data in special situation. For example the outlier filter first filters out data and then the class converts the data to some units. Dou you have any specific problem or thing that you would like to solve?