gandalf15/HX711 #30: Problem with initialization of HC711 in C

cheddarzao, 2020-02-20:
Hello, I have been tried to deal with this HX711, but I have no success. I am new in Raspberry and I want to know how to do that, How to use the "initHX711" (I have been created variables for clock_pin and data_pin, because I have been used a function that gets some parameters in)
I tried to use "read_sacel.c", but it has so much errors of compilation. I tried to run the debug, but no success again, because hasn't compiled and built. I tried to create the "hx" struct, but again no success. I don't know what is wrong with my code, I don't know which is the syntax error. All older issues are starting from who inits with success, what isn't my case. I already downloaded all libraries as needed and all libraries are included in code. Hx711.h, hx711.c and read_scale.c are opened in another pages for verification of syntax.

This is the function that can't compile:
void init_weight(char gpio_clock, char gpio_data)
{
	HX711 hx;
	
	int err = initHX711(&hx, gpio_clock, gpio_data);
	if (err)
	{
		printf("\n!!! Failed to init HX711 struct !!!\n");
	}
}

My Raspberry version: Raspberry 4B
Linux Version: Raspbian GNU/Linux 10 (buster)

HeMe2, 2020-02-21:
Could you please provide the error message from the compiler, to give us a hint what might be wrong?

cheddarzao, 2020-02-21:
> Could you please provide the error message from the compiler, to give us a hint what might be wrong?
> […](#)

Yes, the error message is (I translated from portuguese):

/usr/bin/ld: /tmp/ccOMXwuJ.o: in function "init_weight":
TESTE.c:(.text+0x194): reference not defined to "initHX711"
collect2: error: ld returned 1 exit status
Compilation failed.

The original error message:

/usr/bin/ld: /tmp/ccOMXwuJ.o: na função "iniciar_sensor_peso":
TESTE.c:(.text+0x194): referência não definida para "initHX711"
collect2: error: ld returned 1 exit status
Compilação falhou.

cheddarzao, 2020-02-21:
I already inserted -l hx711 and -l gb_common and I had no success (again).

HeMe2, 2020-02-27:
Hi, 

Firstly I'm sorry that it took me that long to answer. 

The message you posted clearly indicates a linking error. Since the function call to `initHX711` looks correct, I guess that maybe you haven't told your linker where to find the function implementation. The implementation is located in hx711.c, respectively hx711.o after compilation. Your linker needs to know where to find hx711.o. Usually this can be adjusted in the project settings of your IDE. 

Good luck!

cheddarzao, 2020-03-02:
> Hi,
>
> Firstly I'm sorry that it took me that long to answer.
>
> The message you posted clearly indicates a linking error. Since the function call to `initHX711` looks correct, I guess that maybe you [...]

That is the point. I don't know if the call to function is right. I tried to call in different ways, but no success. How do you call in your code, can you show me?

closed 2020-09-30 by gandalf15