gandalf15/HX711 #31: Reading both channels in one script

Raxo96, 2020-02-27:
Hey! :)
I'm working with HX711 module in my master degree in mechatronics and I've noticed that there is some 'bug' in your library. 

When initialising an object HX711 like:`
hx = HX711(dout_pin=5, pd_sck_pin=6)`
u got this at first : 
`def __init__(self,
                 dout_pin,
                 pd_sck_pin,
                 gain_channel_A=128,
                 select_channel='A'):`
and than

`self.select_channel(select_channel)`

this method makes: 

if (channel == 'A'):
            self._wanted_channel = 'A'
        elif (channel == 'B'):
            self._wanted_channel = 'B'`

in result, when I create 4 objects like

hx = HX711(dout_pin=5, pd_sck_pin=6)
 hxb= HX711(dout_pin=5, pd_sck_pin=6)
 hxc= HX711(dout_pin=13, pd_sck_pin=19)
 hxd= HX711(dout_pin=13, pd_sck_pin=19) 
hxd.select_channel(channel='B')
 hxb.select_channel(channel='B')

and two of them set to channel B , and try to read one after another in order
ch A, ch B, ch A, ch B the frequency of reading samples is like 3Hz despite i got my hx711 module set to 80Hz mode.

Why is that?
Because !
`if (channel == 'A'):
            self._wanted_channel = 'A'
        elif (channel == 'B'):
            self._wanted_channel = 'B'`
this library expects that the next reading channel will be the same as currently read.
And because of that, requirement of additional pulses after reading 24 bits are not satisified.
[image]

I think that makes the module reset all the time, and wait to turn on and set again.

I think, that i case like that (reading channels in order ABAB) its enough to do fix like this:(line 383)
if self._wanted_channel == 'A' and self._gain_channel_A == 128:
            if not self._set_channel_gain(**2**):  # send only one bit which is 1
                return False  # return False because channel was not set properly
            else:
                self._current_channel = 'A'  # else set current channel variable
                self._gain_channel_A = 128  # and gain
        elif self._wanted_channel == 'A' and self._gain_channel_A == 64:
            if not self._set_channel_gain(3):  # send three ones
                return False  # return False because channel was not set properly
            else:
                self._current_channel = 'A'  # else set current channel variable
                self._gain_channel_A = 64
        else:
            if not self._set_channel_gain(**1**):  # send two ones
                return False  # return False because channel was not set properly
            else:
                self._current_channel = 'B'  # else set current channel variable

I hope that it will be helpfull for somebody ;)

gandalf15, 2020-09-30:
Thanks for the suggestion. I'll try to improve and implement this in future commits.

Morrious, 2021-06-08:
@Raxo96  I don't think there has been a code update since this issue. I had trouble adapting this code to multiple HX711 sensors, so I made my own repo with an architecture centered around multiple HX711 load cells sharing the same clock/SCK pin.

If you're interested in trying that out:
  https://pypi.org/project/hx711-multi/
  https://github.com/Morrious/hx711-multi

jacktian-web, 2021-08-02:
> 嘿！:)
> 我正在攻读机电一体化硕士学位时使用 HX711 模块，我注意到您的库中存在一些“错误”。
> 
> 当初始化一个对象 HX711 时，比如：` hx = HX711(dout_pin=5, pd_sck_pin=6)`
> 你一开始得到这个：
> `def __init__(self, dout_pin, pd_sck_pin, gain_channel_A=128, select_channel='A'):`
> 然后
> 
> `self.select_channel(select_channel)`
> 
> 这种方法使：
> 
> if (channel == 'A'):
> self._wanted_channel = 'A'
> elif (channel == 'B'):
> self._wanted_channel = 'B'`
> 
> 结果，当我创建 4 个对象时
> 
> HX = HX711（dout_pin = 5，pd_sck_pin = 6）
> HXB = HX711（dout_pin = 5，pd_sck_pin = 6）
> HXC = HX711（dout_pin = 13，pd_sck_pin = 19）
> HXD = HX711（dout_pin = 13，pd_sck_pin = 19）
> HXD。 select_channel(channel='B')
> hxb.select_channel(channel='B')
> 
> 其中两个设置为通道 B ，并尝试依次读取一个接一个，依次为
> ch A、ch B、ch A、ch B，尽​​管我将 hx711 模块设置为 80Hz 模式，但读取样本的频率仍为 3Hz。
> 
> 这是为什么？
> 因为 ！
> `if (channel == 'A'): self._wanted_channel = 'A' elif (channel == 'B'): self._wanted_channel = 'B'`
> 该库期望下一个阅读频道与当前阅读的频道相同。
> 正因为如此，无法满足读取 24 位后附加脉冲的要求。
> [image]
> 
> 我认为这会使模块一直重置，并等待打开并再次设置。
> 
> 我认为，我的情况是这样（按 ABAB 顺序阅读频道）它足以做这样的修复：（第 383 行）
> 如果 self._wanted_channel == 'A' 和 self._gain_channel_A == 128：
> 如果不是 self._set_channel_gain( **2** ): # 只发送一位为 1
> return False # 返回 False 因为通道设置不正确
> else:
> self._current_channel = 'A' # else 设置当前通道变量
> self._gain_channel_A = 128 # 并获得
> elif self._wanted_channel = = 'A' and self._gain_channel_A == 64:
> if not self._set_channel_gain(3): # 发送三个
> return False # 返回 False 因为频道没有正确设置
> else:
> self._current_channel = 'A' # else 设置当前频道多变的
> self._gain_channel_A = 64
> else:
> 如果不是 self._set_channel_gain( **1** ): # 发送两个
> return False # 返回 False 因为通道设置不正确
> else:
> self._current_channel = 'B' # else 设置当前通道变量
> 
> 我希望它会对某人有所帮助;)

Have you done with two channels? My PCB is red, I have no idea which pin would give me an 80 Hz sampling rate.
[image]

Morrious, 2021-08-03:
Your red chip doesn't have a label on it, but based on the circuit board it looks like the orientation of the HX711 is the same as in the picture supplied with this repository. [That picture is here.](https://github.com/gandalf15/HX711/blob/master/rate_pin_80sps.png) You'll note that an obvious matching pin is the DAT pin is connected to the 4th from the bottom right in both pictures.

Please let me know how it goes with the 80Hz, I'm curious what the actual rate ends up being when accounting for some bad returns. I don't run this repository, but I made [my own HX711 repository over here](https://github.com/Morrious/hx711-multi) with an emphasis on sampling multiple HX711 sensors in parallel.

jacktian-web, 2021-08-04:
I am going to give up on this chip. because I am trying to detect high-frequency objects. thank you for your reply.