gandalf15/HX711 #34: How to install?

gregpronko, 2020-03-28:
Hi Gandalf15,

I'm relatively new to Python and RPi so this question is not really an issue but more of a questions. On your readme page there is nothing on how to install your HX711 on a RPi.  Could you update it with some simple instructions on how to install?  And can we install using PIP?  I am using Docker Template on a Balena platform so the ultimate would be to get install instructions that could work in a docker build.

Thanks so much,

Greg

gandalf15, 2020-06-24:
Hi there,
I am sorry for the very late reply but I've been busy. Unfortunately, I am not familiar with Balena platform. It may be possible to do this if you have a direct access to GPIO on the Raspberry PI inside the docker container. If you do any progress please let me know and I would be happy to include your contribution here.

On Raspberry Pi you just clone the repo and run the script. Here is a simple excample how: https://github.com/gandalf15/HX711/blob/master/HX711_Python3/simple_example.py

I'll have to write instructions how to do this step by step and it will be included in the future commits.

Jackfritt, 2020-08-02:
Tried python3 example.py and just got this error

Traceback (most recent call last):
  File "example.py", line 12, in <module>
    err = hx.zero()
  File "/home/pi/src/HX711/HX711_Python3/hx711.py", line 125, in zero
    result = self.get_raw_data_mean(readings)
  File "/home/pi/src/HX711/HX711_Python3/hx711.py", line 446, in get_raw_data_mean
    filtered_data = self._data_filter(data_list)
  File "/home/pi/src/HX711/HX711_Python3/hx711.py", line 680, in outliers_filter
    data_median = stat.median(data)
  File "/usr/lib/python3.7/statistics.py", line 379, in median
    raise StatisticsError("no median for empty data")
statistics.StatisticsError: no median for empty data

Tried to install on pure debian buster raspbian.
Also tried to install 
sudo pip3 install statistics
no help.. Dont know anything about python.....

ilkobg, 2020-08-10:
@Jackfritt Check your HX711 sensor wiring; you probably changed the pins.

MBENKI, 2020-08-11:
Hi @gandalf15 @ilkobg i have the exact same problem with that same output when i run "sudo python3 example.py" in the terminal, here are the pictures of the wiring, i really have no idea if the wires work well, i hope you can see what's wrong from the pictures. Thank you so much.
[image]
[image]
[image]

Jackfritt, 2020-08-11:
I already used the lib from
https://github.com/tatobari/hx711py
There is another pinout in the example files. I used these pins 5 and 6.
You  have to change them in this code...
HX711/HX711_Python3/example.py
...
    #hx = HX711(dout_pin=21, pd_sck_pin=20)
    hx = HX711(dout_pin=5, pd_sck_pin=6)
...

Then the code works as it should....

MBENKI, 2020-08-11:
Thank you very much @Jackfritt  :D , it runs now, (i still get the error message half of the time) although the values i get are not accurate at all, i don't know if the scale is not built well or if this code does not work with load cell (1kg) ... 
[image]

Jackfritt, 2020-08-11:
I use 20kg Sensor and after calibrating routine in example.py it is very accurate (it shows grams). dont know what Temperature and long time measure will differ.... Still testing this..

MBENKI, 2020-08-24:
> I noticed that you don't have a good electrical connection between the HX711 and load cell. Try to solder it properly and this may help. Let us know then.

Yes in fact i did it manually and the connection between the load cell and the hx711 doesn't seem solid, i was searching online to buy those white "connectors" you can see in this video : [https://www.youtube.com/watch?v=Jd3INHU90B0] , English is not my native language that's why i don't know how they're called but if someone have a suggestion then i will take it :)

gandalf15, 2020-08-25:
You can just solder cables directli to pins. This way it is cheaper and easier.

closed 2020-08-25 by gandalf15

MBENKI, 2020-08-28:
Hi everyone so i bought molex connectors and now it is working fine thanks all of you for you help specially @gandalf15 and @Jackfritt  :D