gandalf15/HX711 #39: Crashing after 10s

JeanSolagnier, 2020-12-23:
Dear Gandalf,

I have a problem that the example.py will crash after about 10 seconds. I think it is not reading enough samples and the datalist
 runs out of numbers and cant calculate the median anymore.

I am using a Raspberry pi Zero. HX711 has been modified to 80SPS.
I am using GPIO 5 as Dout and GPIO 6 as SCK.  I also have to write 5 and 6 in the example.py instead of 29 and 31 as per BCM numbering, which I find strange. Can you explain this also?
I really want to use your code since it has the most stable readings compared to others.

Can you help me out?

[image]

JeanSolagnier, 2020-12-23:
I have tried hx._read() in a while loop to see the results and a lot of them come back as "false".
[image]

MilkCrate745, 2021-01-14:
I'm also having the same issue. When i try running the example.py it just crashes after a few seconds with the same error message.

I ran the loop you wrote and there are a lot of false readings. It seems cyclical, there are a few good readings and then a ton of false readings, repeat.

gandalf15, 2021-05-27:
Try now with the latest commit. It should solve the problem.

closed 2021-05-27 by gandalf15