gandalf15/HX711 #43: example.py problem (Attribute error)

TadeuClemente, 2021-08-19:
Hi guys! I'm a beginner in python and raspberry pi, so first i'm sorry for the dumb question. haha

When i run the code, on line 12 appears an attribute error (has no attribute for [err = hx.zero()] ). What might have happened when I installed the HX711 class?

I installed the library normally and the cables are connected correctly. The python version is: 3.7.3.

Please help this poor programming beginner.

Morrious, 2021-08-19:
Is the error saying that "hx" has no attribute? Is it saying there is something within the zero() function that doesn't have an attribute?

The author of this repo is a bit inactive from what I can tell. If you want to give my sister repo with similar functionality and some improvements, you can check it out at: https://github.com/Morrious/hx711-multi

TadeuClemente, 2021-08-19:
The error says this: "AttributeError: 'HX711' has no attribute 'zero'

I'm just trying to get raw data of the strain gauge's electrical resistance variation. 
Would simpler code help? Which wheatstone bridge setup would you recommend?

Morrious, 2021-08-19:
I can't suggest which wheatstone bridge to use, I have been working with scales from China that already had the HX711 wired up.

Your HX711 install must be bad, but it's strange that you're getting to the .zero() line instead of failing at the hx = HX711 line. This to me indicates you have a conflicting HX711 being imported? Here's an example indicating your error and if the class didn't exist:
```python
class TestClass:
    def print_test(self):
        print('test')

testclass = TestClass()
testclass.print_test()
testclass.print_test_bad() # this fails with "'TestClass' object has no attribute 'print_test_bad'"

testclassbad = TestClassDoesntExist()  # this fails with "name 'TestClassDoesntExist' is not defined"
testclassbad.print_test()
```

Morrious, 2021-08-19:
You know what, you might not be importing the right thing. This repo doesn't have a pip install, so you have to clone the code locally and import it there. I bet you ran `pip3 install hx711`, which is not this code. That's something written by someone else entirely and has very little functionality: https://pypi.org/project/hx711/

TadeuClemente, 2021-08-19:
I actually used 'pip3 install HX711'. I didn't know it wasn't the right way. 

Tomorrow I will install your HX711_multi by pip and try to run.
Which example do you recommend for raw data?

Morrious, 2021-08-19:
> Which example do you recommend for raw data?

In the example I include in my readme, you can see the below line which includes raw values for your sensors after some averaging if you input a value greater than 1. I found that my sensors often fail to return data every time, and often return bad data. These samples are filtered based on their bit values and their deviation from median to get much cleaner data. The "raw" values are after this filtering and averaging.
```python
raw_vals = hx711.read_raw(readings_to_average=10)
```

If you want the really really raw data, you can access that within the load_cell data after performing the read above:
```python
for lc in hx711._load_cells:
    print(lc.raw_reads) # these are the bit-shifted values read from the 24-but hx711
    print(lc.reads) # these are the raw values after being converted to signed integers
```

TadeuClemente, 2021-08-31:
Hi, Morris! 

Now, when i try to run the code with HX711_multi, the raspbi returns the error message: "ModuleNotFoundError: No module named 'hx711_multi'."

Morrious, 2021-08-31:
hmmm, I haven't worked with a Pi Zero, but it should work. What do you get when you run `pip3 show hx711_multi`?