gandalf15/HX711 #45: Reading linearly increases/decreases

henryminhtran, 2021-11-29:
Hi, recently I ran into a problem where after starting and input the calibration weight, the first output reading was the closest to the actual value but then gradually increases/decreases over time. I spotted a linear relationship between each reading but have no idea what was going on and how to fix this. Can anyone help?

Morrious, 2021-11-29:
How much is the drift over time? You may just be seeing mechanical drift over time from the load cells. https://www.arlynscales.com/industrial-scales/what-causes-load-cell-drift/

henryminhtran, 2021-11-29:
Thank you for the prompt response! I'll look into that.

henryminhtran, 2021-11-30:
The drift over time doesn't seem to stop and I have no idea nor experience to conduct the tests mentioned in the provided links. I'm running the load cells the first time in order to calibrate it. Please help!

Morrious, 2021-11-30:
Do the measurement values still change when you add or subtract weight? As in the measurements are still working properly but drifting slowly?

The way I use HX711 measurements is I post-process with a running average and take measurements by tracking large deltas in the measurements. This accounts for the drift over time because a small delta from drift doesn't get tracked.

If your drift is constant, you could use numpy to calculate the slope and rotate your data, but that gets a bit more complicated.

Morrious, 2021-11-30:
I don't work with this repo anymore as I needed much more parallel sampling functionality. If you have time and want to switch to a different module, mine is over at [hx711-multi](https://github.com/Morrious/hx711-multi) and is pip installable. I can work together with you on post processing code if you're interested over there.

henryminhtran, 2021-11-30:
> 

Yes the drift still continues disregarding the adding/subtracting weight. The started value was correctly says I put a 250g bag of rice of in, then the value will read 250g within the first 2s and then slowly drifting away with 2g per second and doesn't seem to stop. Last time I ran the program it still continue at 500g, which is painfully wrong.

henryminhtran, 2021-11-30:
@Morrious That'd be really nice and helpfuly of you. I'm working on a group design project and we hit dead end if the load cell can't seem to work properly. I'll try it out and let you know for sure so thank you so much for hanging around

henryminhtran, 2021-11-30:
For the mean time, here's the datasheet for the load cell that I'm currently using: https://theloadcelldepot.com/wp-content/uploads/2014/06/DS_cheap_load_cell.pdf. My setup consists of 4 of the mentioned load cells wired into a Junction Box to ideally sum up into one analog signal before connecting to the HX711 and send the digital signal to the Rasp Pi. The load cells are rated for 10V excitation voltage but I read somewhere online saying supplying 5V voltage from the RPi should be fine to get it to work with relatively decent accuracy. Please feel free to let me know otherwise