gandalf15/HX711 #49: AttributeError: 'HX711' object has no attribute 'zero'

snakepatricia, 2022-12-27:
Hi community,

My goal is to get through the code step by step. So I want to see what each line does when running, to get better understanding.
From the example.py code there was a command hx.zero().
Tried to run it but could not succeed. Just got the error AttributeError: 'HX711' object has no attribute 'zero'
I know there is an open issue with the same problem, but still no solution so I considered to try again.

The pins are correctly connected to the raspberry 3 and the library also got installed like described. The problem also occurs to other commands like hx.get_raw_data_mean() and hx.get_data_mean()
Seems like the installation failed somewhere...

Appreciate any help!

Morrious, 2022-12-27:
The error you're getting indicates that the "hx" object just doesn't recognize a "zero()" function. Are you sure you have the correct hx711 module installed and imported?

snakepatricia, 2022-12-28:
@Morrious Thank you for your answer!
I installed it with: pip3 install 'git+https://github.com/bytedisciple/HX711.git#egg=HX711&subdirectory=HX711_Python3'

Already saw it and honestly tried to install it with pip3 but it could not be found. 
Maybe the raspberry itself is the problem. I'll try both installations with another one, maybe it will work. :)

Morrious, 2022-12-28:
@snakepatricia I doubt that this specific error means there's something wrong with the Pi other than the installs. Have you tried uninstalling hx711 and reinstalling with that pip command?

'pip3 uninstall hx711'